nccoos/platforms #47: Upgrade one 150 kHz ADCP (sn 15373) to hold four (4) batteries

neaptide, 2016-11-17:
Current status of Harvey's 150 kHz ADCP (sn 15373) is that is holds two (2) battery packs.  

Lots of questions:
- How do we pay for this tube upgrade if cost is ~$10k? We did not budget for this in PEACH?
- Do we have an additional external battery pack holder that holds two (2) to complement the two (2) in the ADCP?  (I thought we got one of these with 150 kHz purchase but can't remember.) And will the pod accommodate this configuration?

- [ ] get quote to upgrade tube to hold four (4) batteries
- [ ] determine source of funds to purchase
- [ ] get requisition since >$5k and write sole source justification
- [ ] place order

---------------------------------------
Trip said in an email dated June 6, 2016, "It's probably worth mentioning that your ADCP battery case only holds two batteries while ours hold four.  For a longer deployment time it would probably be required to get a larger battery tube (~$10k each)."

muglia, 2016-11-22:
Does it make sense for me to actually purchase this added battery pack? I think running the sole source through UNC should be on your end, have it shipped here, and we'll equip. it for deployment, no?

hseim, 2016-11-22:
Don’t think it matters too much, but if I need to do a sole source it is probably easier for me to initiate the purchase.  Do we have a quote?

H

neaptide, 2016-11-22:
No quote.  Waiting on answers to the questions (revised for clarity) ...

neaptide, 2017-01-13:
If I read the quote from TRDI, it looks like four (4) battery packs (WHABPLR) come with this purchase.

hseim, 2017-01-13:
Excellent. Mike told me it may take a different style of battery than the smaller cases.

H

muglia, 2017-01-13:
Great, and yes, we confirmed that it looks like the same battery packets with different mounting holes.
Mike

neaptide, 2017-01-24:
Requisition has been approved and Elise is emailing purchasing agent to start order.

neaptide, 2017-02-21:
PO sent Feb 9.  Expected ship date is Mar 1 2017 (Fedex Gnd) to Chapel Hill.  Will need to get it to CSI after we receive it.

desimonenm, 2017-03-07:
Any update on the Battery pack shipment? I see it is suppose to be shipped on the 1st.  Has it been shipped and is there an estimated delivery date to Chapel Hill?  Knowing this would give me a better idea of what kind of time constraints I will have when installing the pack and calibrating the moorings.

neaptide, 2017-03-07:
Received in Chapel Hill 20 min ago.  

Trying to figure out the logistics to get it from here to CSI.  Stand by ...

closed 2017-03-07 by neaptide