nccoos/platforms #64: Freight Logistics to WHOI

neaptide, 2017-02-21:
Make arrangements to truck buoys, ADCP pods to WHOI by March 30, 2017.  This will require a full truck load (FTL) Step-down trailer.  The route is Morehead City, NC>Wanchese, NC>Woods Hole, MA.  Working with Ryan Transportation (Andrew Seiler).  

- [ ] Schedule dates for pickup, stop-over and delivery to arrive March 30.
- [ ] Verify height to top of buoy ring (or highest point) when buoy is on side
- [ ] List all pieces, dimensions as loaded, weights
- [ ] Layout of pieces
- [ ] Determine extra space available for additional pallet boxes
- [ ] Get metal strapping tools and materials to secure anchor to pallet -- IMS or purchase?
- [ ] Get shipping box from storage for delicate parts taken off buoys for transport

Number | Description | Weight (lbs) | Dimensions (feet, LxWxH)
----------| -------------| -------------|-------------------
1 | Buoy (B1) | 2000 | 12'x7'x9.5' on side (*estimate*)
2 | Buoy (B2) | 2000 | 12'x7'x9.5' on side (*estimate*)
3 | Small Bottom Frame (B1) | 200 | 5'x5'x3'
4 | Small Bottom Frame (B2) | 200 | 5'x5'x3'
5 | Anchor Assembly | 2800 | 3’x’3’x3’
6 | Large Bottom Pod (A5) | 1200 | 6'x6'x5' (*estimate*)
7 | Large Bottom Pod (A6) | 1200 | 6'x6'x5' (*estimate*)

There might be a pallet or two extra of parts (still trying to figure this out) that we can tuck into empty spaces.  Total weight not to exceed 20,000 lbs, but more like 12,000 lbs.   We have equipment and personnel to load and off-load at each location.  

[image]

muglia, 2017-02-22:
I'd like to fill in the table, but not seeing how to do it so...
#6 A5 $120,703
#7 A6 $108.640 (less the hydrophone value)

neaptide, 2017-03-09:
Shipment schedule finalized
Tues, March 28 at 8AM -- First pickup at IMS, Morehead City, NC (Stacy Davis)
Tues, March 28 at 3-4PM -- Second pickup at CSI in Wanchese, NC (Mike Muglia)
Thurs, March 30 by Noon or so -- Drop off at WHOI in Woods Hole, MA (Brian Hogue)

closed 2017-03-09 by neaptide