nccoos/platforms #79: Order Zinc Anodes

neaptide, 2017-11-07:
For the buoy:
- [ ] (4) plate anodes (2 per buoy) on buoy frame
- [ ] (24 ??) bolt-end anodes (12 per buoy??) on buoy frame
- [ ] (2) prop-shaft collar anode for each mooring swivel
- [ ] (??) various others 

For the bottom frames:
- [ ] (4) flat-circular (trim tab and rudder) 
- [ ] (??) collars for cross arms
- [ ] (6) bolt-end anodes for bolts holding pod feet
- [ ] pencil anode for acoustic releases (??) hard to find may have to order from ORE

 Can get most from West Marine Pro account

hseim, 2017-11-08:
I checked my notes and it appears I bought 1" ID zinc collars for the bottom frame cross bars.  They were a bit large but we put copper wire in to fill the gap.

neaptide, 2017-11-08:
keeping this open until we have all ordered anodes in hand

neaptide, 2017-11-08:
Thanks @hseim.  I edited [main comment ](https://github.com/nccoos/platforms/issues/79#issue-271906009)for this issue.

neaptide, 2017-12-19:
most all ordered thru West Marine, being shipped to Chapel Hill before Friday Dec 22